RobotLocomotion/drake #23125: [ci] Add experimental unprovisioned equivalents of experimental provisioned jobs

tyler-yankee, 2025-06-24:
The list of [experimental unprovisioned jobs](https://github.com/RobotLocomotion/drake/blob/jenkins-jobs-experimental/request-jobs-unprovisioned.txt) should be expanded to be a superset of the list of [experimental provisioned jobs](https://github.com/RobotLocomotion/drake/blob/jenkins-jobs-experimental/request-jobs-provisioned.txt), so that changes to the prereqs can be tested on the full production suite.

tyler-yankee, 2025-06-24:
@jwnimmer-tri @BetsyMcPhail This issue really has two stages, noting from our Slack discussion that some of these unprovisioned jobs actually do exist, but we're just not updating `request-jobs-unprovisioned.txt` for the desired use case.

1. Add the actual unprovisioned jobs that are missing. By my count, this is essentially just the special dev-centric flavors:
```
linux-noble-unprovisioned-clang-bazel-experimental-address-sanitizer
linux-noble-unprovisioned-clang-bazel-experimental-coverage
linux-noble-unprovisioned-clang-bazel-experimental-everything-address-sanitizer
linux-noble-unprovisioned-clang-bazel-experimental-everything-leak-sanitizer
linux-noble-unprovisioned-clang-bazel-experimental-everything-thread-sanitizer
linux-noble-unprovisioned-clang-bazel-experimental-everything-undefined-behavior-sanitizer
linux-noble-unprovisioned-clang-bazel-experimental-everything-valgrind-memcheck
linux-noble-unprovisioned-clang-bazel-experimental-leak-sanitizer
linux-noble-unprovisioned-clang-bazel-experimental-thread-sanitizer
linux-noble-unprovisioned-clang-bazel-experimental-undefined-behavior-sanitizer
linux-noble-unprovisioned-clang-bazel-experimental-valgrind-memcheck
linux-noble-unprovisioned-gcc-bazel-experimental-coverage
```
i.e., all the other jobs in `request-jobs-provisioned.txt` (aside from special ones like cache server health check) already have an unprovisioned equivalent, that we're just not currently listing out. (e.g., I think this would have been useful for #23080?)

2. Sync up the `request-jobs-*.txt` lists to make it easier for developers with prerequisite changes. I'm open to suggestions here, but my proposal here is that we:
    * merge the existing files into some `request-jobs-experimental-production.txt` (maybe comments to separate provisioned and unprovisioned sections)
    * add some new `request-jobs-unprovisioned-prereq-changes.txt` file that lists the entire unprovisioned production-as-experimental suite. This differs from the unprovisioned section of the above in that it would additionally contain the unprovisioned equivalents of the provisioned experimental jobs that come from production.

BetsyMcPhail, 2025-06-25:
I confirmed the list of noble jobs. I also double-checked that the missing jammy jobs are just missing from the `request-jobs` file, but already exist in Jenkins.

closed 2025-07-15 by tyler-yankee