RobotLocomotion/drake #23161: Memory leak when calling Discrete...Event

Kongx231, 2025-07-03:
### What happened?

In our codebase, we construct new simulation diagrams in a loop to apply different scene conditions for domain randomization. We have noticed after `Drake 1.37.0` our memory has continuously increased and our programs/computers crash. We can reproduce this issue both with `pip install drake` and our internal Amazon version of drake.

We have traced one of our issues (unsure if there is more) to having a discrete initialization event `DeclareInitializationDiscreteUpdateEvent`. We have provided a minimally reproducible example where memory usage increases after each loop. It is visible using `memory_profiler` python package.

[image]

Steps to recreate this plot
```bash
python -m mprof run test_memory_mre.py
python -m mprof plot -o mprof_memory_usage --title "Connected and discrete initialized"
```

```python
# test_memory_mre.py
from pydrake.all import (
    Simulator,
    RobotDiagramBuilder,
    LeafSystem,
)
import gc
import logging
import numpy as np

USE_DISCRETE_INIT = True

class NullSystem(LeafSystem):
    def __init__(self):
        LeafSystem.__init__(self)
        # Critical for the leak is having the information piped at least once.
        # Without this, no memory leak.
        if USE_DISCRETE_INIT:
            self.DeclareInitializationDiscreteUpdateEvent(self.DiscreteInit)

    def DiscreteInit(self, context, discrete_state):
        self._data = np.zeros((1000, 1000))

def main():
    logging.getLogger("drake").setLevel(logging.ERROR)
    for i in range(9999):
        print(f"Running iteration {i}, {gc.get_count()}")
        builder = RobotDiagramBuilder(time_step=0.05)
        plant = builder.plant()
        plant.Finalize()
        builder.builder().AddSystem(NullSystem())
        diagram = builder.Build()
        context = diagram.CreateDefaultContext()
        # Create Simulator
        simulator = Simulator(diagram, context)
        simulator.Initialize()
        gc.collect()

if __name__ == "__main__":
    main()
```

### Version

1.42.0

### What operating system are you using?

Ubuntu 22.04

### What installation option are you using?

pip install drake

### Relevant log output

```shell
Running iteration 0, (224, 10, 1)
Running iteration 1, (0, 0, 0)
Running iteration 2, (0, 0, 0)
Running iteration 3, (0, 0, 0)
...
Running iteration 9998, (0, 0, 0)
```

rpoyner-tri, 2025-07-08:
There are several issues I'm trying to untangle here:
* I don't currently think any of the NullSystem code is necessary to result in a (~20MiB/sec) leak
* the program as offered (`diagram` is still reachable when `gc.collect()` is called) confounds garbage collection more than I would have thought. Why are not (apparently?) unreachable objects from a prior iteration not reaped by GC?

Here is a version of the test program I am currently experimenting with:
```python
# test_memory_mre.py
from pydrake.all import RobotDiagramBuilder
import gc
import logging
import time

def main():
    logging.getLogger("drake").setLevel(logging.ERROR)
    start_time = time.time()
    for i in range(20000):
        if i % 10 == 0:
            print(f"Running iteration {i}, {gc.get_count()},"
                  f" {time.time() - start_time}")
        builder = RobotDiagramBuilder(time_step=0.05)
        diagram = builder.Build()
        # Uncomment the line below to stop leaking.
        # del builder, diagram
        gc.collect()

if __name__ == "__main__":
    main()
```

With the `del` line commented out, the program leaks ~20MiB/sec. With it executing, the program does not leak. The main thing that the `del` line does is to ensure that the `builder <-> diagram` cycle is unreachable when `gc.collect()` runs.

It is important to recognize that python does not create a variable scope for many constructs that look like blocks: `for, if, else, with`. So without the `del` statement, objects created in iteration `i==0` will still be reachable until nearly the end of iteration `i==1`. Still, I would expect the old objects to be reaped in later `gc.collect()` calls. If that were happening, the memory graph would be mostly flat, with a slightly higher maximum. Of course, that is not what we are seeing.

I will continue to investigate this, but perhaps the above discussion is enough to give you a work-around for your larger program.

rpoyner-tri, 2025-07-08:
I think the leak can be explained by details of the garbage collector's behavior, though I will not (yet) claim to fully understand it.

Adding a bit more logging:
```python
        if i % 10 == 0:
            print(f"Running iteration {i}, {gc.get_count()},"
                  f" g0:{len(gc.get_objects(0))},"
                  f" g1:{len(gc.get_objects(1))},"
                  f" g2:{len(gc.get_objects(2))},"
                  f" garbage:{len(gc.garbage)},"
                  f" {time.time() - start_time}")

```
We can now see the sizes of the "generations" lists, and of the "garbage" list.

In a run with the `del` statement commented out, the number of objects hanging around in generation 2 climbs steadily:
```
Running iteration 0, (33, 5, 9), g0:128, g1:3056, g2:97333, garbage:0, 0.003450632095336914
Running iteration 10, (0, 0, 0), g0:0, g1:0, g2:100163, garbage:0, 0.10377168655395508
Running iteration 20, (0, 0, 0), g0:0, g1:0, g2:100193, garbage:0, 0.18248939514160156
Running iteration 30, (0, 0, 0), g0:0, g1:0, g2:100223, garbage:0, 0.25713610649108887
Running iteration 40, (0, 0, 0), g0:0, g1:0, g2:100253, garbage:0, 0.341778039932251
Running iteration 50, (0, 0, 0), g0:0, g1:0, g2:100283, garbage:0, 0.4179115295410156
Running iteration 60, (0, 0, 0), g0:0, g1:0, g2:100313, garbage:0, 0.49518847465515137
Running iteration 70, (0, 0, 0), g0:0, g1:0, g2:100343, garbage:0, 0.5701978206634521
Running iteration 80, (0, 0, 0), g0:0, g1:0, g2:100373, garbage:0, 0.6510624885559082
Running iteration 90, (0, 0, 0), g0:0, g1:0, g2:100403, garbage:0, 0.7264456748962402
```

In a run with the `del` statement executing, the size of the generation 2 list levels off:
```
Running iteration 0, (33, 5, 9), g0:128, g1:3056, g2:97333, garbage:0, 0.0036802291870117188
Running iteration 10, (0, 0, 0), g0:0, g1:0, g2:100130, garbage:0, 0.09395384788513184
Running iteration 20, (0, 0, 0), g0:0, g1:0, g2:100130, garbage:0, 0.16175413131713867
Running iteration 30, (0, 0, 0), g0:0, g1:0, g2:100130, garbage:0, 0.23100996017456055
Running iteration 40, (0, 0, 0), g0:0, g1:0, g2:100130, garbage:0, 0.2991316318511963
Running iteration 50, (0, 0, 0), g0:0, g1:0, g2:100130, garbage:0, 0.3685629367828369
Running iteration 60, (0, 0, 0), g0:0, g1:0, g2:100130, garbage:0, 0.4376657009124756
Running iteration 70, (0, 0, 0), g0:0, g1:0, g2:100130, garbage:0, 0.5068581104278564
Running iteration 80, (0, 0, 0), g0:0, g1:0, g2:100130, garbage:0, 0.582200288772583
Running iteration 90, (0, 0, 0), g0:0, g1:0, g2:100130, garbage:0, 0.6498792171478271
```

In the leaking scenario, we have forced a full scan while the objects of interest are still reachable. They cannot be collected while they are in the youngest generation, and so end up in the oldest generation.  Because the size of the oldest generation can grow without bound, the GC design tries to avoid scanning the whole thing at once, and tries to rate limit those scans. Somehow, this has the aggregate effect of failing to collect some unreachable cycles (ours).

There is not much the `pydrake` layer can do to avoid such situations, while simultaneously avoiding too-short lifetime hazards (seg faults, bogus assertions, etc.).

rpoyner-tri, 2025-07-08:
Found the problem. It's specific to RobotDiagram/Builder, the ordinary Diagram/Builder abstractions don't suffer from it.

It turns out that we are tripping on some sort of address/object reuse pattern that allows the internal bookkeeping of pydrake's ref cycle mechanism to leak from one iteration of the loop to later ones 😱 . Since the internals just accumulate refs in a `set()`, the problem snowballs.

Detail: pybind11 keeps track of c++/python object correspondences, based on addresses, to support "casting". When we have the bad luck that some generation of a c++ object is allocated at the exact same address as a predecessor, and that predecessor's python object is still alive, then `pybind11::cast()` of the c++ pointer can return the "old" python object, along with all of it's bookkeeping data. Running RobotDiagram/Builder in a tight loop allows the internal DiagramBuilder to be allocated at the same address, which allows its bookkeeping to leak, which effectively creates an ever-growing graph of references to all the generations of diagrams.

In the variations above that didn't leak, `del diagram` allowed the iteration's cyclic objects to be reaped right away, so the address-aliasing didn't have any stale bookkeeping to propagate.

PR #23171 (2025-07-16): [pydrake] Fix a RobotDiagramBuilder memory leak
Closes: #23161

---
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/23171)
closed 2025-07-16 by jwnimmer-tri