RobotLocomotion/drake #23183: Support common GLTF extensions/ avoid RuntimeError due to missing visual extensions

nepfaff, 2025-07-14:
**Is your feature request related to a problem? Please describe.**
I keep getting warnings/ runtime errors when loading GLTF files that use extensions.

Example 1:
```
RuntimeError: error: ERROR: In vtkGLTFDocumentLoaderInternals.cxx, line 1331: vtkGLTFDocumentLoader (0x58048918ec00): glTF extension KHR_texture_transform is required in this model, but not supported by this loader. Aborting
```

Example 2:
```
Warning: In vtkGLTFDocumentLoaderInternals.cxx, line 1354: vtkGLTFDocumentLoader (0x58048852f9a0): glTF extension KHR_materials_clearcoat is used in this model, but not supported by this loader. The extension will be ignored.)
```
same for `KHR_materials_specular`, `KHR_materials_ior`, etc.

These should only affect rendering functionality. Hence, the RuntimeError seems a little extreme.

**Describe the solution you'd like**
It would be nice if Drake could parse arbitrary GLTF files. The runtime errors are particularly annoying.

I wouldn't mind if Drake did not throw an error, use worse VTK/ meshcat rendering, but maintain these properties for rendering with Drake Blender Server.
I'd assume that it shouldn't be too hard to ignore these properties, but keep them in the GLTF files that get sent to the server.

**Additional context**
I'm working on importing online assets into Drake.

jwnimmer-tri, 2025-07-14:
Anything listed in a glTF file's `extensionsRequired` is literally required in order to open the file.  If the file is supposed to be load-able without requiring that extension, then the file should list the extension in `extensionsUsed` but not in `extensionsRequired`.

In terms of suppressing warnings (for extensions that are "used" but not "required"), the `RenderEngineVtkParams` config struct has a field named `gltf_extensions` which can be used to silence the logged warnings.

The very latest tip of VTK upstream supports `KHR_texture_transform` (within the past few weeks).  See #23144 for our pending upgrade PR.  I'm not sure if the version in progress there is quite new enough to get the new feature, but in any case once that PR lands, it should be easy to upgrade again to the very latest and get the new feature.

jwnimmer-tri, 2025-07-14:
Plausibly, Drake could offer an additional config flag to suppress raising the exception, something like this:

```patch
--- a/geometry/render_vtk/render_engine_vtk_params.h
+++ b/geometry/render_vtk/render_engine_vtk_params.h
@@ -88,6 +88,13 @@ struct GltfExtension {
    all unimplemented extensions will log a warning, but users can configure
    specific extensions to be quiet. */
   bool warn_unimplemented{true};
+
+  /** Whether to raise an error when this extension is required (i.e., is listed
+   in `extensionsRequired`) but isn't implemented by the render engine. By
+   default, all required-but-unimplemented extensions will raise an error, but
+   users can opt-in to ignoring the file's metadata; be aware that the rendered
+   images will almost certainly be wrong (and often, severely so). */
+  bool error_when_required_and_unimplemented{true};
 };
 
 /** Construction parameters for the RenderEngineVtk.  */
```

If @SeanCurtis-TRI agrees with the idea, I think we would accept a PR that added this, but I don't think we'll do it ourselves.  The work-around mentioned above is the correct solution -- if users want to ignore missing extensions, they can say that in the glTF file(s) directly.

nepfaff, 2025-07-14:
These insights are great. Thank you very much!

It seems fair to throw an error for `extensionsRequired` then. I will modify my code to delete that field from my GLTF files.

I'd be happy to close that issue for now, as the extensions seem to be pending upstream VTK instead of Drake support. 

Upgrading VTK to support `KHR_texture_transform` would be amazing, but I guess it looks like this is on the roadmap anyway?