RobotLocomotion/drake #23190: SchunkWsgTrajectoryGenerator cannot be properly reset with the internal `trajectory_`

hongkai-dai, 2025-07-14:
### What happened?

`SchunkWsgTrajectoryGenerator` class has an `trajectory_` field https://github.com/RobotLocomotion/drake/blob/5fbb89e6e380c418b3f651ebde22a8f9203b6b1e/manipulation/schunk_wsg/schunk_wsg_trajectory_generator.h#L114. When we reset the context, this `trajectory_` is not set properly. The retained `trajectory_` causes unexpected output of the class.

As discussed with @jwnimmer-tri , we should add a state that controls initializing the `trajectory_` object. Then when we reset the context, this updated state will trigger initializing `trajetory_`.

### Relevant log output

```shell

```

hongkai-dai, 2025-07-15:
I think when I do reset, I don't want to call `SchunkWsgTrajectoryGenerator::UpdateTrajectory()` which sets `trajectory_` to a new value; instead I would like to set `trajectory_` to null, same as when the `SchunkWsgTrajectoryGenerator` object was constructed.

So how does the system know that I am resetting it? One candidate approach is that I add an abstract state `bool set_traj_to_null` to `SchunkWsgTrajectoryGenerator`. This abstract state is set to `true` initially. Then when we update the state in the simulation, for example when we call `CalcDiscreteUpdate()`, if `set_traj_to_null` is true, then we set `trajectory_ = nullptr`, and then change `set_traj_to_null` to `false`.

In this way, when I call `simulator.get_mutable_context().SetTimeStateAndParametersFrom(context_reset)`, where `context_reset` is a clone of the simulator context at the beginning of the previous simulation, it will set `SchunkWsgTrajectoryGenerator::trajectory_` to nullptr.

@jwnimmer-tri WDYT?

jwnimmer-tri, 2025-07-15:
`CalcDiscreteUpdate` cannot change abstract state.

I think a simpler solution is to take one of the 4 existing discrete state variables, and have its default value be something other than zero.  For example, default the `last_target_position` to infinity.  That can be the signal that a reset is needed on the first step, and `CalcDiscreteUpdate` will naturally change it to a new value.