RobotLocomotion/drake #23191: MultibodyPlant provide HasDeformableModel

hongkai-dai, 2025-07-15:
Currently when I have a MultibodyPlant object, I don't know if it has a deformable model. I can call `MultibodyPlant::deformable_model()`, and it throws an exception if it doesn't have one. It would be better to provide a function
```cc
bool MultibodyPlant::HasDeformableModel() const;
```

xuchenhan-tri, 2025-07-15:
MultibodyPlant is supposed to be add a DeformableModel upon construction. Could you show a code snippet of how you get the throw?

hongkai-dai, 2025-07-15:
The documentation of `MultibodyPlant::deformable_model()` says so https://github.com/RobotLocomotion/drake/blob/6ce9719857522b50e431ad91f27bab197ca6adb2/multibody/plant/multibody_plant.h#L2633-L2640

xuchenhan-tri, 2025-07-15:
This documentation should have been updated in #21357 but it slipped through the cracks.

xuchenhan-tri, 2025-07-15:
Yes, that's correct.

closed 2025-07-15 by jwnimmer-tri