RobotLocomotion/drake #23204: Restore subheading hyperlinks in pydrake API documentation

tyler-yankee, 2025-07-21:
With #23052 (see also #23078) we now build documentation on Noble, which involved a new version of sphinx. This  removed the hyperlinks to individual functions on the pydrake API that were there previously.

The [sphinx docs](https://www.sphinx-doc.org/en/master/usage/configuration.html#confval-html_permalinks) seem to indicate that `html_permalinks = True` and `html_permalinks_icon = '¶'` are the default values, but possibly those are being overridden somewhere, and from an initial investigation it seems setting them explicitly in [conf.py](https://github.com/RobotLocomotion/drake/blob/master/doc/pydrake/conf.py) doesn't work either.

tyler-yankee, 2025-07-23:
I've started looking into this. Some scratch notes:

* When the call to `doc.pydrake.pydrake_sphinx_extension` is removed from `conf.py` entirely, obviously all the styling gets messed up, but the links are there. 
* Tried removing a bunch of stuff, notably Drake's customizations in `doc/pydrake/pydrake_sphinx_extension.py`, to see if I could narrow down a place where the default setting was being overridden or somehow hidden with styling. This was to no avail.
  - CSS
  - `pydoc.PythonDomain.directives['template'] = pydoc.PyClasslike`
  - `patch(autodoc.ClassLevelDocumenter, 'resolve_name', patch_resolve_name)`, and the two lines below
* I saw that importing the `napoleon` extension tends to be prickly with other extensions including `autodoc`, but reordering didn't affect anything (or in some cases broke the build).
* I briefly looked at `autodoc_default_flags = ['members', 'private-members', 'special-members']`.

Finally, I confirmed by inspecting the generated html of the site on Jammy and Noble that the `<a>` and `<href>` tags are present on the Jammy version and not on the Noble version.

tyler-yankee, 2025-08-11:
I don't have any ideas beyond running in a VM or container, save for seeing if there is a method of installing the newer Sphinx on Jammy from non-apt. I was running on a Noble machine.

mwoehlke-kitware, 2025-08-25:
FYI, it *looks* like Sphinx on Noble *does* spit out a ton of `WARNING: error while formatting signature for ...: 'NoneType' object has no attribute 'groups'` that look related to the silent failure of `domains.python.PyObject.handle_signature`. (They don't show up on Jammy, or with use of `IrregularExpression` removed.) Unfortunately, these are buried under many, many *other* warnings from Sphinx, which is probably why they weren't noticed.

We should probably consider not ignoring Sphinx warnings...

On a related note, has anyone attempted to upstream this functionality? I suspect Drake isn't the only project that might benefit.

closed 2025-08-29 by SeanCurtis-TRI