RobotLocomotion/drake #23209: [ci] Iron out nightly job scheduling

tyler-yankee, 2025-07-22:
Currently, nightly jobs are scheduled as:

* 1am: all packaging and wheel jobs
* 2-4am: the rest of the jobs (spread out randomly based on a hash of the job name)

As discussed in #23207, there are two possible improvements here:

* We'd like to maintain the priority of the packaging and wheel jobs, but can it be implemented more intelligently? Something like the [Priority Sorter plugin](https://plugins.jenkins.io/PrioritySorter/) seems viable.
* If possible, let's move the coverage jobs to start earlier (1am?), as they take really long to run and occasionally disrupt CI maintenance if we're waiting for them to finish. (In general, it's easier for buildcops -- particularly on Eastern Time -- if all the nightly jobs are done before working hours.)

BetsyMcPhail, 2025-07-22:
For reference, jobs are currently assigned priorities using the referenced plugin. Views are used sort jobs.

Priorities are assigned top-down by the first match:

2. Packaging
3. Documentation
4. Pull Request
5. Experimental
7. Continuous
6. Nighty
1. Staging

Note that Wheels and Coverage currently aren't prioritized.

tyler-yankee, 2025-07-23:
Thanks for pointing this out! Erm, I'm seeing:

1. Staging
2-5. (Above)
6. Nightly
7. Continuous

The plugin only allows for 7 unique priority levels. We could conceivably leave coverage unprioritized but start it at an earlier time, and then prioritize wheels at the same level as packaging (2).

BetsyMcPhail, 2025-07-23:
Ugh, GitHub helpfully renumbered the list for me... Hopefully this looks better

2 Packaging 
3 Documentation
4 Pull Request
5 Experimental
7 Continuous
6 Nighty
1 Staging