RobotLocomotion/drake #23302: [multibody] Floating base body should be restricted to World frame

sherm1, 2025-08-21:
### Background
MultibodyPlant provides special treatment of "free bodies" (any body with a 6 dof "floating" joint) and "floating base bodies" (free bodies whose parent body is World). There was much confusion about these two concepts which was heroically untangled by @SeanCurtis-TRI in #21808.

I'm having to revisit this as part of #23245 since bodies welded to World can simply be part of World now rather than separate mobilized bodies.

Per f2f with Sean, the intent for singling out floating base bodies as a concept is 
(1) they can be posed arbitrarily like any free body, and
(2) the coordinates of their floating joints can be interpreted directly as World coordinates.

In order to achieve property (2) the floating joint's parent frame must be the World frame. Looking in the code, I see that we did not require that. Instead, we just require that the parent frame is _any_ frame attached to the World body. **Sean thinks this is a bug**.

### Proposal
Tighten up the rules so that `is_floating()` is only true for a free body if the parent frame is the actual World frame, not just any frame on World.

### Discussion
This revised definition would remain unchanged when #23245 merges bodies onto World. Merging bodies onto World reduces them effectively to just frames on the World body. Hence to be consistent with the current implementation I would have to extend the definition of floating base body to permit the inboard frame to be on one of the merged bodies. That's certainly possible but seems awkward, not to mention a really bad idea if the current implementation is just a bug.

This would be a breaking change if anyone is depending on the current implementation. However, all our automatically-created floating base joints use the World frame as the parent frame. In order for this to cause trouble, a user would have had to
- add an explicit floating joint between a body and a frame on World that isn't the World frame, and
- make use of the bespoke floating base body API in MultibodyPlant (rather than going through the Joint API).

I believe that is extremely unlikely so we could make the definition change with minimal pain. However, I could be wrong and would like to hear if anyone thinks this is a real concern.

cc @rcory @RussTedrake @calderpg-tri

jwnimmer-tri, 2025-08-21:
Ah.  It took me many re-readings of the post to fine the key detail.  I'll underline it for others who might also struggle. The key point is "body" vs "frame.

The current documentation of `is_floating()` already says that the parent must be the world **body**:

```c++
  /// (Advanced) Returns `true` if this body is granted 6-dofs by a Mobilizer
  /// and the parent body of this body's associated 6-dof joint is `world`.
  /// ...
  bool is_floating() const {
```

The proposal here is to require that it's literally the world **frame** instead, i.e., newly disallowing free joint's parent being a fixed-offset frame on the world body.

---

This seems easy to smoke test for impact -- add the requirement and run the full regression test suite for Drake, Anzu, and Russ's two courses.

sherm1, 2025-10-20:
Fixed by #23412 

The new definition of a floating base body is one that is on an _automatically added_ 6dof joint. Those joints are always added between the body frame and the world frame, so that eliminates the problem described in this issue.

closed 2025-10-20 by sherm1