RobotLocomotion/drake #23440: macOS Sonoma (14) support end of life

jwnimmer-tri, 2025-09-16:
**Is your feature request related to a problem? Please describe.**

Per https://drake.mit.edu/stable.html: "Drake intends to support the two most recent versions of Ubuntu LTS and macOS on an ongoing basis."  Now that Tahoe (26) is out, we will no longer be supporting Sonoma (14) moving forward.

**Describe the solution you'd like**

Remove Sonoma from our documentation, notate its removal in our end of life documentation, turn off CI for it (switching its packaging jobs to Sequoia, etc).  Then clean up the code (removing any Sequoia compat hacks or fixing "once we have Sequoia" TODOs).

**Additional context**

See #20511 one PR from the prior removal.

tyler-yankee, 2025-09-18:
Drake can build Sequoia [wheels](https://drake-jenkins.csail.mit.edu/job/mac-arm-sequoia-clang-wheel-experimental-release/job/master/1/) and [binary packages](https://drake-jenkins.csail.mit.edu/job/mac-arm-sequoia-clang-cmake-experimental-packaging/job/master/1/) -- not that this was unexpected, but we never even had experimental variants of these before.

BetsyMcPhail, 2025-09-22:
@tyler-yankee #21978 has a checklist that was used when removing Ventura support

tyler-yankee, 2025-10-03:
Sonoma AMIs deleted, and Jenkins has two Sequoia Macs up and running for now (until Tahoe support is added).

closed 2025-10-03 by tyler-yankee